olofk/edalize #411: GHDL Support for cocotb simulation flow

psi-chuffine, 2024-01-18:
I use GHDL with cocotb and am working to move to fusesoc.
It looks like the cocotb flow currently only supports Icarus and modelsim.

Based on the cocotb docs it doesn't look too difficult to extend:
https://docs.cocotb.org/en/stable/custom_flows.html#ghdl

maybe something like 
```
"ghdl": (
                    "run_options",
                    ["--vpi", "`cocotb-config --lib-name-path vpi ghdl`"],
                ),
```
Patched in https://github.com/olofk/edalize/blob/18972c4ce508fa98748a01011773e3508c8302d4/edalize/flows/sim.py#L23

I haven't dug deep enough into the code to know what else may need to be modified.

psi-chuffine, 2024-01-19:
This does not seem to be as simple as I had hoped, the ghdl.py in /tools/ is very broken. I'm working on that at the moment.

olofk, 2024-01-21:
Sorry about that. I remember having some headaches trying to get the ghdl backend working. Since I'm not a regular ghdl user myself, it's harder to see if I have done some weird things there. Fixes are highly appreciated.