olofk/edalize #417: QSys Pro 23 files are not processed correctly by the filter

GCHQDeveloper137, 2024-03-08:
I've recently had to create some new QSys files, and was surprised to see the build failed because the IP didn't get generated correctly. This was a bit of a surprised, as an existing project withb QSys in it still worked fine with the latest FuseSoC and Edalize.

On investigation, it seems that for v23, they've completely changed the format of the XML that QSys generates. After tracing through the code, it's become clear that the check inside qsys_file_filter in quartus.py is now failing to detect that this is a QSys Pro file, and so not running generation on it. To make life more complicated, they still call it version 1.0, so detecting it can't be a simple version check inside the file.

I think it should be fine to check both variants. I'll try some local changes, and if they work will put up a PR to work around this file format change. I see the fallback just assumed Standard rather than Pro - I suspect changing this to assume the same as the version of Quartus installed might be a better choice, so I might do this at the same time, unless anyone objects.

GCHQDeveloper137, 2024-03-08:
Aha, the fallback comment states that we should treat it as standard, but looking at the code it actually gets treated as the /opposite/ of the variant of the tool in use. That's clearly a very strange thing to do - if running standard the template will ignore this, and if running pro it would be filtered out so would ignore it! I'll make it default to the version in use, and update the comment to match the code.

GCHQDeveloper137, 2024-03-08:
Aha, the old behaviour was correct - we're matching versions up in here, my mistake. I'll just update the comments to be a little clearer.

olofk, 2024-03-08:
EDA tool vendors never cease to amaze me when it comes to finding new ways to create compatibility with the world around them, including their own tools.

As I don't have access to Quartus Pro myself, a PR would be most welcome if there is a fix needed.

GCHQDeveloper137, 2024-03-08:
I've got something that works fine for all the tool versions I have access to (old Standard, old Pro and new Pro). Thankfully, after updating the tests it spotted an issue with what you'd expect the new format Standard to look like, so I've fixed that. PR incoming shortly.

closed 2024-03-11 by GCHQDeveloper137