olofk/edalize #419: Jinja2 issue witth edarool loader

CCCVega, 2024-03-26:
There seems to be an issue with the loaders in edatools.py that has been driving me crazy for days. 

I've tried multiple versions of edalize, fusesoc, and jinja but nothing seems to work. 

 File "/usr/local/bin/fusesoc", line 11, in <module>
    load_entry_point('fusesoc==0.5', 'console_scripts', 'fusesoc')()
  File "/usr/local/lib/python3.6/dist-packages/fusesoc/main.py", line 775, in main
    fusesoc(args)
  File "/usr/local/lib/python3.6/dist-packages/fusesoc/main.py", line 765, in fusesoc
    args.func(cm, args)
  File "/usr/local/lib/python3.6/dist-packages/fusesoc/main.py", line 336, in run
    args.verbose,
  File "/usr/local/lib/python3.6/dist-packages/fusesoc/main.py", line 435, in run_backend
    backend = backend_class(edam=edam, work_root=work_root, verbose=verbose)
  File "/usr/local/lib/python3.6/dist-packages/edalize/edatool.py", line 183, in __init__
    loader=PackageLoader(__package__, "templates"),
  File "/usr/local/lib/python3.6/dist-packages/jinja2/loaders.py", line 291, in __init__
    assert loader is not None, "A loader was not found for the package."
AssertionError: A loader was not found for the package.

olofk, 2024-03-27:
I have seen this problem popping up a couple of times in other places, and I have absolutely zero idea of what is going on and I'm not able to reporoduce it here. I would very much like to find a solution to this but don't know where to start.

olofk, 2024-03-27:
Maybe you could try to just change `__package__` to Edalize on line 183 in edatool and see if that helps?

DVCoder, 2024-04-16:
I'm also running into a similar issue. Please let me know if anyone has a fix/workaround.

DVCoder, 2024-04-16:
Found a fix that worked for me. I was also using Python 3.6 and I upgraded to Python 3.9 and that fixed it for me.