olofk/edalize #423: Vivado flow not re-run when Verilog parameter is changed

keegandent, 2024-04-05:
With the way the current Makefile is written, Synthesis, Implementation, etc. are not re-run when a Verilog parameter is changed in the `edam`. This is because the `.xpr` file is an [order-only prerequisite](https://www.gnu.org/software/make/manual/html_node/Prerequisite-Types.html) of the successive targets, and is the only file updated when the parameter is changed.

I'm really not a makefile expert. My gut tells me one of two solutions needs to be used here. One is having design-affecting parameters in their own `.tcl` file that is a normal prerequisite of the synth target (or have it be part of the existing `_synth.tcl` which is already a prerequisite there). The other is using some sort of phony target, but I can't quite see a justification for that when there is a material code change to prompt the rebuild.

I'd be willing to help with working towards a solution here, but I'm wondering if there aren't other gaps where changes in the design need to be added to this issue and we can try to solve them all at once.

keegandent, 2024-04-05:
Additionally, updating the source files has no effect because those are also only normal prerequisites for the `.xpr` file, meaning the `_synth.tcl` script is never run so even if `source_mgmt_mode` is set to `DisplayOnly` or `All`, Vivado never has the opportunity to check if sources are out of date.

olofk, 2024-04-05:
Hmm... unfortunately it looks like you're right. I'm too tired right now to figure out the best way forward but it sounds like you're on the right track. IIRC, I had to make the xpr file an order-only dep because stupid Vivado keeps updating it everytime it launches. There's is probably a way to properly fix this though. Putting the parameters in a separate tcl file is worth investigating, but as you mention, there might be other issues too.

keegandent, 2024-04-06:
> Hmm... unfortunately it looks like you're right. I'm too tired right now to figure out the best way forward but it sounds like you're on the right track. IIRC, I had to make the xpr file an [...]

All good boss, honestly you deserve massive thanks for the time you put into this project. Given how PITA Vivado expects to be in charge of dependency checking, maybe the simple approach is to just let it take control “Project Flow” instead of trying to have the makefile do it? I’m not sure how that will play out with using it for only *part* of a flow (e.g. using yosys for synthesis). But at least the all-Vivado flow would work as well as (and as poorly as) the underlying tool itself.

Lucas-KB, 2025-08-20:
I have tested changing this and it does seem to work I have submitted a Pull Request https://github.com/olofk/edalize/pull/472